thelsing/knx #287: Via ETS: Programming the physical address only works very rarely. Mostly never. At some point, with luck, it works.

radontec, 2024-07-26:
Hello everyone.
I have a question.
I have the example knx-demo.ino working in Platform IO.
knx-demo-ip.knxprod works.

I have already made my own project.
Not much changed. Just adapted the channels. But everything is still on a demo basis.

Via ETS: Programming the physical address only works very rarely. Mostly never. At some point, with luck, it works.
Then I can transfer the application program over and over again. Always works perfectly. No matter how often, it works great.

I delete the ESP. Then I can no longer get the device programmed via ETS With a lot of patience and luck, it works at some point. Can't say what the problem is.

Can anyone tell me where I could look? What information do you need? Possibly my project?

[image]

[image]

thelsing, 2024-07-26:
Its hard to say what the problem is. You probably have compare the telegrams ETS sends and receives with the one the devices sends and recieves. 
People were having issues with igmp-proxies. Maybe that's your problem too.

The hex codes are hard to read. It might be good to add a toString() method to the CemiFrame class and print them out in `DataLinkLayer::frameReceived` and `IpDataLinkLayer::sendFrame` (it would be even better to add it to `DataLinkLayer::sendFrame` but this requires a bit more work, as `DataLinkLayer::sendFrame` is currently a pure virtual method.

Trexis5, 2024-07-26:
`sent packages in last 1000ms: 0 curTime: 262
<- 06 10 05 30 00 11 29 00 B0 E0 FF FF 00 00 01 01 40 
-> 06 10 05 30 00 11 29 00 B0 C0 11 67 00 00 01 01 00 
KnxIpServiceType: 530
RoutingIndication
AckType: 0
AddressType: 128
Destination: 0
Source: 4455
FrameFormat: 128
Priority: 0
NPDU: 0x3ffb1d10
Own Address: 65535
SystemBroadcast: 16
AddressType is GroupAddress and Destination is 0
SystemBroadcast is not SysBroadcast
Destination Address: 0
Source Address: 65535
Address Type: 128
Priority: 0
System Broadcast: 16
NPDU Octet Count: 1
sent packages in last 1000ms: 0 curTime: 293
<- 06 10 05 30 00 11 29 00 B0 E0 FF FF 00 00 01 01 40 
-> 06 10 05 30 00 13 29 00 BC D0 11 52 18 96 03 00 80 00 64 
KnxIpServiceType: 530
RoutingIndication
AckType: 0
AddressType: 128
Destination: 6294
Source: 4434
FrameFormat: 128
Priority: 12
NPDU: 0x3ffb1d10
Own Address: 65535
SystemBroadcast: 16
AddressType is not GroupAddress or Destination is not 0
-> 06 10 05 30 00 11 29 00 B0 C0 11 67 00 00 01 01 00 
KnxIpServiceType: 530
RoutingIndication
AckType: 0
AddressType: 128
Destination: 0
Source: 4455
FrameFormat: 128
Priority: 0
NPDU: 0x3ffb1d10
Own Address: 65535
SystemBroadcast: 16
AddressType is GroupAddress and Destination is 0
SystemBroadcast is not SysBroadcast
Destination Address: 0
Source Address: 65535
Address Type: 128
Priority: 0
System Broadcast: 16
NPDU Octet Count: 1
sent packages in last 1000ms: 0 curTime: 323
<- 06 10 05 30 00 11 29 00 B0 E0 FF FF 00 00 01 01 40 
-> 06 10 05 30 00 13 29 00 BC D0 11 57 20 96 03 00 80 00 64 
KnxIpServiceType: 530
RoutingIndication
AckType: 0
AddressType: 128
Destination: 8342
Source: 4439
FrameFormat: 128
Priority: 12
NPDU: 0x3ffb1d10
Own Address: 65535
SystemBroadcast: 16
AddressType is not GroupAddress or Destination is not 0
-> 06 10 05 30 00 11 29 00 BC D0 11 57 20 97 01 00 80 
KnxIpServiceType: 530
RoutingIndication
AckType: 0
AddressType: 128
Destination: 8343
Source: 4439
FrameFormat: 128
Priority: 12
NPDU: 0x3ffb1d10
Own Address: 65535
SystemBroadcast: 16
AddressType is not GroupAddress or Destination is not 0`

Trexis5, 2024-07-27:
Here the Debug:
`--------------------------------------------
CemiFrame Received: 
AckType: 0
AddressType: 128
Destination: 12358
Source: 4417
FrameFormat: 128
Priority: 4
NPDU: 0x3ffb1d10
Own Address: 65535
SystemBroadcast: 16
29 00 B4 D0 11 41 30 46 03 00 80 38 00 
CemiFrame Received: 
Header: 29 00 B4 D0 11 41 
Msg Code: 30
Add.Info Length: 46
Ctrl 1: | 03 (Frame Type: 0, Reserved: 0, Repeat Flag: 0, System Broadcast: 0, Priority: 0, Acknowledge Request: 1, Confirm: 1)
Ctrl 2: | 00 (Destination Address Type: 0, Hop Count: 0, Extended Frame Format: 0)
Source Address: 8038 (32824)
Dest. Address: 0000 (0)
Data Length: 00
APDU: 00 00 
--------------------------------------------`

thelsing, 2024-08-30:
I added more logging in the devel branch. If you set the ApplicationLayer Logger to Info you should be able to spot the difference between ets monitor and the telegrams that reach the device.