thelsing/knx #290: Compile error on knx_facade.h when importing library in third party PIO project

PaoloTK, 2024-08-12:
I would like to use this library to add KNX compatibility to WLED, which is a software to drive LEDs that runs on ESP. However when I try to include <knx.h> on my class I get an error:

> expected type-specifier before ')' token

The error occurs on line 80 col 74 of knx_facade.h, on the B init:

https://github.com/thelsing/knx/blob/150f3fc24b363e5c35ca474eb8332ee82a262d0d/src/knx_facade.h#L80

I haven't started using the library yet, the include alone causes the compile error. 

Here's my platformio.ini:

```
[env:esp32_knx]
extends = esp32dev
platform = espressif32
framework = arduino
board = esp32dev
lib_deps = ${esp32.lib_deps}
  https://github.com/thelsing/knx
upload_speed = 921600
lib_ldf_mode = deep+
build_flags = ${common.build_flags} ${esp32.build_flags}
  -D USERMOD_KNX
  -D WLED_DEBUG
  -D MASK_VERSION=0x07B0
  -Wno-unknown-pragmas
```

Any idea if this is an issue on my side, WLED side or the library?

thelsing, 2024-08-13:
Without seeing any code it's hard to say. Can you try to build the knx-demo example first? According to github action CI it compiles fine for esp32. If you have the code somewhere I can take a look at it.

thelsing, 2024-08-30:
No further reply: Closing.

closed 2024-08-30 by thelsing