thelsing/knx #293: Invalid TPuart frames

michaelkusolitsch, 2024-08-30:
hi,
Every Frame i receive on the rp2040 from the ncn5121 is invalid. Therefore no communication with the device is possible
e.g.: "Invalid:  00.08.001 -> 08.00.002 [I_______] ( BC 08 01 80 02 45 13 08 01 80 02 45 4F )"

The hardware should work fine as another program of mine works fine with the setup. (hardcoded tpuart test)

MASK_VERSION=0x07B0
platformio library version 1.5.0

thelsing, 2024-08-31:
Did you try the knx-demo project? 
Try those build-options in  

build_flags =
  -DMASK_VERSION=0x07B0        ; Normal Device TP1 = Twisted Pair (cable)
  -DKNX_FLASH_SIZE=0x8000      ; default = 0x1000 = 4KiB; must be multiple of 4096
  ;-DKNX_FLASH_OFFSET=0xF4000  ; default = 0x180000 = 1.5MiB; must be multiple of 4096
  -DUSE_TP_RX_QUEUE            ; Improved performance on TP by using a queue instead of immediate frame processing
  -DPIO_FRAMEWORK_ARDUINO_ENABLE_RTTI  ; according to rp2040_arduino_platform.cpp: RTTI must be set to enabled in the board options
  -DDEBUG_TP_FRAMES
  -Wuninitialized
  -Wunused-variable
  -Wno-unknown-pragmas

Those work for @embed-me at #291. I have no rp2040 so I can't check myself.

michaelkusolitsch, 2024-09-01:
Yes i try to use the knx-demo project.
i could fix the invalid message errors when using the dma options of the rp2040 platform. It looks like bytes where lost before without the dma option.

  -DUSE_KNX_DMA_UART=0
  -DUSE_KNX_DMA_IRQ=0
  -DMAX_TX_QUEUE=100
  -DMAX_RX_QUEUE_BYTES=1024

now tpuart error occurre when trying to write the individual address with ets
"TPUart state error: 1000000"

is there a description what the individual bits of the error state mean?

But as far as i could see while debugging the "address read" from ets (before writing) is handled correctly and a response is beeing generated. Unfortunately i can't tell if the response has the right information in the datagram or if the write is not possible due to the TPUART errors.

michaelkusolitsch, 2024-09-05:
i could trace down the problem but i don't have a solution yet. As i couldn't see the response in ets, i monitored the tpuart between the rp2040 and the ncn5121 with an oscilloscope. There i could see that there will ococure framing errors after 3 or 4 bytes on the sending side at the rp2040 of the uart communication. Therefore the TPUART errors are displayed since no valid data is received at the ncn5121. So i think it is a problem in the rp2040 arduino platform.

michaelkusolitsch, 2024-09-05:
i found the problem. When creating no global knx instance i can set the serial and all settings my self which works perfectly. But the dma settings are fixed. so when setting the serial to 8 bit no parity, it doesnt effect the even parity setting of the dma in the platform. After changing that it workes!

closed 2024-09-05 by michaelkusolitsch

thelsing, 2024-09-05:
Thanks for reporting back.

Ing-Dom, 2024-09-06:
@traxanos 
fyi