thelsing/knx #294: knx-linux-coupler fails on download application

ajlajlajl, 2024-09-30:
Hi,
So I've been trying to setup the knx-linux-coupler in my ubuntu, its connected to a knx development board (NCN5120).
It works alright when I send write/read commands or while monitoring the bus for packets that devices exchange between eachother.
But when I try to download an application on a device its tries for a sec and then fails with: "Failed to connect to 1.1.6" , with 1.1.6 being the device that Im trying to download app on.
I tried to watch the packets on wireshark but it dowsnt look like theres any request that has not been responded. and Im not really familiar with the underlying knx stack so Im not sire what the probmen is.
btw I enabled some of debug outputs and this is the output when download fails:
```
pRxF: [1] 01.01.006 -> 01/00/057 [__R_D__A] ( 9C 11 06 08 39 E3 00 80 14 20 12 )
Inbound:  01.01.006 -> 01/00/057 [__R_D__A] ( 9C 11 06 08 39 E3 00 80 14 20 12 )
  TP<: 9C 11 06 08 39 E3 00 80 14 20 12 
  CEMI<: 29 00 9C E0 11 06 08 39 03 00 80 14 20 
ToTunnel L_data_ind: src: 1106 dst: 839 frame: 29 00 9C E0 11 06 08 39 03 00 80 14 20 
Send to Channel: 4
L_data_req: src: FF01 dst: 1106 frame: 11 00 B0 60 FF 01 11 06 00 80 
L_data_con: src: FF01 dst: 1106 frame: 2E 00 B0 60 FF 01 11 06 00 80 
Send to Channel: 4
ToTunnel L_data_ind: src: FF01 dst: 1106 frame: 29 00 B0 60 FF 01 11 06 00 80 
Found no Tunnel for IA: 1106
L_data_req: src: FF01 dst: 1106 frame: 11 00 B0 60 FF 01 11 06 01 43 00 
L_data_con: src: FF01 dst: 1106 frame: 2E 00 B0 60 FF 01 11 06 01 43 00 
Send to Channel: 4
ToTunnel L_data_ind: src: FF01 dst: 1106 frame: 29 00 B0 60 FF 01 11 06 01 43 00 
Found no Tunnel for IA: 1106
L_data_req: src: FF01 dst: 1106 frame: 11 00 B0 60 FF 01 11 06 00 81 
L_data_con: src: FF01 dst: 1106 frame: 2E 00 B0 60 FF 01 11 06 00 81 
Send to Channel: 4
ToTunnel L_data_ind: src: FF01 dst: 1106 frame: 29 00 B0 60 FF 01 11 06 00 81 
Found no Tunnel for IA: 1106
```

this "Found no Tunnel for IA: 1106" message is all I get and Im not sure what it means..
Do I need to set some config or something to make this works? since it looks like to be a internal logic error.
or am I doing something wrong?

Thanks.

thelsing, 2024-10-01:
I guess I broke this on a merge. I'll try to look into this this weekend.

thelsing, 2024-10-01:
Which knxprod did you use?

aliglzr, 2024-10-02:
I have this issue too
What branch or version I need to use for a bug free coupler?

thelsing, 2024-10-02:
@aliglzr Which knxprod do you use?

ajlajlajl, 2024-10-03:
> Which knxprod did you use?

Im using the knx-linux-coupler. And im trying to download on a different device in tp network through this.

aliglzr, 2024-10-07:
@thelsing any good news?

aliglzr, 2024-10-12:
@thelsing
I'm sorry to message you again
But my project depends on your answer to solve the problem, can you help me or solve my problem?
Your help is very valuable to me

thelsing, 2024-10-12:
You should be able to program other devices without tunnelling but with multicast. Just by using the network interface as the programming interface in ETS. If you want to use tunneling try changing -DKNX_TUNNELING to -DKNX_TUNNELING=1 at
https://github.com/thelsing/knx/blob/7280cd0e5362d028065c571388253f85df544638/examples/knx-linux-coupler/CMakeLists.txt#L16

You can also try to remove it to disable tunnelling.

aliglzr, 2024-12-08:
Any updates?
I am still stuck on this issue...

zwjin1210, 2025-09-04:
I tested the same situation, how should I deal with it?

aliglzr, 2025-09-04:
I couldn’t find any solution for this.