thelsing/knx #295: Example not working

labmaster, 2024-10-03:
I pulled https://github.com/thelsing/knx/tree/master/examples/knx-demo-diy and built it (PlatformIO) without errors.

Uploaded it to a ESP32 (NodeMCU ESP32) that has been connected to a MicroBCU on Uart2 (Serial2 / GPIO16 and GPIO17)
Additional i hooked up a LED (with Current lim. Resistor) and a Button (with Pullup) to GPIO18 (LED) and GPIO19 (BUTTON)

When i reset the System i get the following:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
readMemory
RESTORED FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
DataObject api changed, any data stored in flash is invalid.
expexted DataObject api version: 1, stored api version: FFFF
ETS has to reprogram PA and application!
TP is disconnected
```

When i press the button the LED toggles on/off and i get 
"progmode on" and "progmode off" in the serial (Serial1) monitor.

But the system does not connect to the KNX Bus.
I already hooked up a logic analyzer (to RXd and TXD) and can see the MicroBCU (NCN5120) sending Data (19200, 8N1E) to the ESP GPIO16 but the ESP32 does nothing on the TXD Pin.

In the Example the only parts that i change are the Button/LED definitions. Should there be anything additional changed added that is not documented there ?

I'am a bit lost now how to go further.

labmaster, 2024-10-04:
Found the problem.

In the Example code the Pin Definitions for UART2 are missing:
at line:
Serial2.begin(19200);  // KNX, pin 16,17 on EPS32

add the Pin Definitions, like:
Serial2.begin(19200, SERIAL_8E1, 16, 17);  // KNX, pin 16,17 on EPS32

dzungpv, 2024-10-21:
@labmaster I have problem with this example, ETS could not programing it. When I send program it show: "Please press programing button" but I have press button to change to "progmode on".
Logs:
```
readMemory
RESTORED FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
DataObject api changed, any data stored in flash is invalid.
expexted DataObject api version: 1, stored api version: FFFF
ETS has to reprogram PA and application!
TP is disconnected
TP is connected
progmode on
```