thelsing/knx #297: Compatible with NCN5120/NCN5121

dzungpv, 2024-10-17:
Do this lib compatible with NCN5120/NCN5121? I read the #2 seem it not working fine.

betamoojw, 2024-10-17:
Yes, it is compatible with NCN5120/NCN5121 from one of my practical exercises. This library has evolved and been refined since its birth, with many excellent contributors.

dzungpv, 2024-10-17:
> Yes, it is compatible with NCN5120/NCN5121 from one of my practical exercises. This library has evolved and been refined since its birth, with many excellent contributors.

OK, I see in OpenKNX repo they using device with an BCU module using NCN5120/5130 and this lib to make application. https://github.com/OpenKNX/OpenKNX/blob/main/accessories/MicroBCU2/doc/MicroBCU2_Schematic_V00.15.pdf

closed 2024-10-21 by dzungpv