thelsing/knx #303: I can only access upto 255 group objects in System B 07B0.

anand-010, 2024-12-17:
I am trying to build a ir transmitter using this library.
My plan is to add upto 1000 ir commands but when i tried to use 256 or more from the group object it frozen the code.
```
        for (uint8_t channel = 0; channel < 254; channel++)
        {
            knx.getGroupObject(channel + 1).callback(Ir_callback);
        }
        for (uint8_t channel = 0; channel < 254; channel++)
        {
            knx.getGroupObject(channel + 1).dataPointType(DPT_Trigger);
        }
```
it worked until this point when i am trying to access the 256th object it frozen.
do you have any idea ?.
while looking into the knx specification System B 07B0 supports up to 65536 group objects.
i am using a rp2040 based board with 16MB flash.
```
	-DKNX_UART_RX_PIN=17
	-DKNX_UART_TX_PIN=16
	-DMASK_VERSION=0x07B0
	; -DKNX_FLASH_SIZE=16384
	-DKNX_FLASH_OFFSET=0xE00000  ;Place at 14MB offset
	-DKNX_FLASH_SIZE=0x100000  ;1MB for KNX data
	-D PIO_FRAMEWORK_ARDUINO_ENABLE_RTTI
	-DKNX_BUTTON=19
	-DKNX_LED=18
	-DKNX_LED_ACTIVE_ON=1
```

thelsing, 2024-12-17:
The maximum values are defined in the knxprod file. Did you change them there?

anand-010, 2025-01-27:
No i haven't changed anything in kaenx crator in the Application -> General  tab Addressable max entities is 65535.
is there anything i need to check in the kaenx creator ?.