thelsing/knx #305: Support for the STKNX chip

caiservass, 2025-01-23:
Hi all,

I wanted to check in to see if there have been any recent updates regarding the compatibility of the STKNX chip with this library. Is there any progress, or are there any planned updates or forks addressing this?

Additionally, I’m looking to test the STKNX chip with a board like the Raspberry Pi Pico, but I’ve been unable to find any schematics or documentation on how to connect the chip to a dev board.

If anyone has experience with this setup or can provide a schematic or guidance on connecting the STKNX chip to the Pico (or any other board), it would be greatly appreciated.

Thanks in advance!

jeff25, 2025-01-26:
Which value you See in stknx? It s a analog Chip ans a lot of Code need to Implemented in Code.

pesimon, 2025-01-30:
Not sure if you found that already https://github.com/mistay/stknx ...