thelsing/knx #307: Ota to new version from 2.1.0, lost the address and data (ESP32)

dzungpv, 2025-03-08:
I have some running version from master branch commit https://github.com/thelsing/knx/commit/bfe5ffd91f3bfb7f281ce515b9128efeace88f7b.
It is version 2.1.0.
Yesterday I build the firmware and take ota update, it is esp32 board and it lose the programing address and probably all the data.

dzungpv, 2025-03-08:
I fixed it by revert the apiVersion here: https://github.com/dzungpv/knx/commit/8767a60f79bff4245733865d5d6ea9d772188e7e, but it is not good solution. Proper solution is check the version and in the flash and make change when ota update, now it failed to read back when hard-code version change.

thelsing, 2025-03-09:
This came from the openknx merge. As nothing changed it the stack I'll just revert it back.

closed 2025-03-09 by thelsing